j8sr0230/Nodes #3: Roadmap?

luzpaz, 2022-08-12:
Roadmaps help people understand what the intention of this workbench is in the long term and how to get there. It also signals tasks other devs can be invited to take on and contribute.

luzpaz, 2023-03-18:
@j8sr0230 has development stopped on this workbench ?

j8sr0230, 2023-03-19:
@luzpaz Sorry for replying so late. No, I will continue to develop FreeCAD Nodes! I had very little time in January and February due to an important project deadline. But this project is completed now. Now I'm on vacation for a few days and after that I'll have again more time for FreeCAD Nodes.

Therefore I am currently reworking the MVC pattern with the intention to represent the data model with a QStandartItemModel and derive all needed views from it. I want to translate the node graph model into a [dask graph](https://docs.dask.org/en/stable/graphs.html) which can be processed completely parallel and asyncron. 

Do you know if FreeCAD's Qt binding implements QML (Qt Modeling Language)? I would like to use this new technology for the datamodel and views?

luzpaz, 2023-03-19:
No we aren't using QML. I asked yorik this and his response was essentially QML becomes highly specialized and the preference is to keep using more mainstream Qt approaches so the barrier for entry for devs is kept higher.

michacassola, 2023-06-07:
Hey @j8sr0230, how are you doing? Everything ok?

yorikvanhavre, 2023-06-08:
My understanding is that QML widgets are programmed, and not designed with a tool like QDesigner anymore. So this is better for programmers (you have way more possibilities with QML) but less simple for non-programmers. So I think the "core" UI system of FreeCAD should still rely as much as possible on standard widgets so non-programmer people can easily plug into it and extend.

That said, there is nothing that would prevent people from using QML for specific things in FreeCAD. I think this nodes 
module for example, would be a very good case for it.

There have been already experiments with QML in FreeCAD... There has been an early stub at a rendering workbench in FreeCAD some centuries ago, that used QML. But I can't find it back :/

luzpaz, 2023-06-10:
@yorikvanhavre you mean the @ickby [FreeCAD UI overlay experiments](https://forum.freecad.org/viewtopic.php?t=23053&sid=f6e4591015ac10b3be2d43bf2ce41f88) ?

yorikvanhavre, 2023-06-12:
Found it! It was a rendering module developed by @mrlukeparry in wow.. 2012 https://forum.freecad.org/viewtopic.php?p=21815&hilit=rendering+workbench#p21815 IIRC it used some QML. But I don't know if the code still exists :sweat_smile:

luzpaz, 2024-12-18:
soft bump