j8sr0230/Nodes #5: Add pyqt-node-editor to allowed pip list in FreeCAD

luzpaz, 2022-08-17:
Hey @chennes would if be OK if we added https://gitlab.com/pavel.krupala/pyqt-node-editor.git as an accepted 3rd party pip dependency so we can add this repo as an addon to FreeCAD-addons ?

chennes, 2022-08-18:
I don't have any objection, feel free to add it to the allowed packages list.

luzpaz, 2022-08-19:
chennes, I haven't looked at the code but will the addon manager be able to appropriately find and download pyqt-node-editor

chennes, 2022-08-19:
As long as pip can find it: which typically means it has to be in pypi -- in this case, here: https://pypi.org/project/nodeeditor/

luzpaz, 2022-08-22:
Dependencies have been added to upstream permission list via https://github.com/FreeCAD/FreeCAD-addons/pull/232

closed 2022-08-22 by luzpaz