j8sr0230/Nodes #6: Add to the Addon Manager

luzpaz, 2022-08-19:
#5  
https://github.com/FreeCAD/FreeCAD-addons/issues/231  
https://github.com/FreeCAD/FreeCAD-addons/pull/232  

Automate the rest of the [installation(https://github.com/j8sr0230/fc_nodes#installation) via the Addon Manager

luzpaz, 2022-09-03:
@FlachyJoe what else needs to happen besides physically adding this to Addon Manager ?

FlachyJoe, 2022-09-03:
You can already have it in your Addon Manager 
[image]

and all users will see it soon: https://github.com/FreeCAD/FreeCAD-addons/issues/231

luzpaz, 2022-10-28:
Awkward Arrays dependency has been approved as a 3rd party python library in the FreeCAD Addon Manager: https://github.com/FreeCAD/FreeCAD-addons/pull/241

luzpaz, 2022-10-28:
Is there anything left for FC Nodes to install seamlessly from Addon Manager?

j8sr0230, 2022-10-28:
No, everything there! The installation worked without problems:) Just tested it with the latest daily:

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.30716 (Git)
Build type: Release
Branch: master
Hash: cbd19631428f1d9eb4faae30220551608a648589
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.3
  * Curves 0.5.0
  * DynamicData 2.46.0
  * fasteners 0.3.46
  * fc_nodes 0.1.0
  * lattice2 1.0.0
  * MyWorkbench
  * ThreadProfile 1.81.0
```
[image]

luzpaz, 2022-10-28:
Sweet! Lets update the README then ?

closed 2022-10-28 by luzpaz