j8sr0230/Nodes #19: Improve socket type

FlachyJoe, 2022-09-07:
Hi,
in order to keep socket typed, we need to deal with nodes which accept some types in a list, for example Part_Extrude can get Wire or Face as base.

I think about this solution but didn't dig in the code:
```
from enum import Flag, auto

class ComparableFlag(Flag):
    def __eq__(self, other):
        if not(isinstance(other, ComparableFlag)):
            raise NotImplementedError()
        return ((self & other).value == self.value) or ((self & other).value == other.value)

class SocketType(ComparableFlag):
    FLOAT   = auto()
    INT     = auto()
    STRING  = auto()
    # ...
```

then Part_Extrude input socket type could be `SocketType.WIRE | SocketType.FACE`

FlachyJoe, 2022-09-08:
I've reflexed about this a bit more and I eventually think use string is better because node writer could easy add his own type and we could also use wildcard for whatever-you-want sockets.

j8sr0230, 2022-09-09:
@FlachyJoe Can I close this after merging [https://github.com/j8sr0230/fc_nodes/pull/21](https://github.com/j8sr0230/fc_nodes/pull/21)?

FlachyJoe, 2022-09-09:
Close with #21

closed 2022-09-09 by FlachyJoe