j8sr0230/Nodes #83: Nodes Icons

Emilien-Etadam, 2022-11-09:
Hi, 
I can help to put an icon for each node (reuse existing one from freecad or create a new one) let me know if this is needed.

I am an advanced user of Solidworks and Grasshopper so I following this Addon with a lot of hope. 
I am not a programmer but I would like to help.

Thanks.

luzpaz, 2022-11-15:
@j8sr0230 bump

j8sr0230, 2022-11-16:
@Emilien-Etadam @luzpaz Sorry it took me so long to reply, but job, family, kids etc .... :) Great that you would like to contribute to my project. You don't have to be a programmer to support me (by the way, I am "only" a mechanical engineer)! 

The context menu for creating new nodes can be accessed by right-clicking in the scene graph. In this menu, each node should have its own icon. Currently, the same icon is displayed as a placeholder for each node. You are welcome to contribute to inserting or creating the node icons. For this purpose, we should always adopt the icon corresponding to the FreeCAD function if it already exists (see [FreeCAD Artwork](https://wiki.freecadweb.org/Artwork)). If none exists yet, you can let your creativity run wild. Please keep in mind that the icons should work with both dark and light FreeCAD style sheets. You could also use Grasshopper 3D, Sverchock or Geometry Nodes as template.

[image]

To contribute, copy your new icon (32x32) into the icons folder (1) and link to it in the source code of the desired node (2, 3). If everything works in your fork, create a pull request and I'll take care of the merging.

I hope that was understandable. If you still have questions, just contact me. Welcome to the team and have fun helping to create FreeCAD Nodes icons :)

Emilien-Etadam, 2022-11-16:
[image]

It works !

j8sr0230, 2022-11-16:
Great, immediately looks more professional ;)

Emilien-Etadam, 2022-11-16:
[image]

Yep !

Emilien-Etadam, 2022-11-16:
[image]

j8sr0230, 2022-11-17:
Thanks for the support, looks really great!

Emilien-Etadam, 2022-11-18:
[image]
 4 New Icons

luzpaz, 2022-11-18:
FANTASTIC!!!

Emilien-Etadam, 2022-11-21:
> 

Ok let me check how to do this the right way...

Emilien-Etadam, 2022-11-21:
[image]
4 New Icons

luzpaz, 2022-11-21:
CC @mistermakernl (just a heads up that this is happening and you might like to keep it on your radar)

j8sr0230, 2022-11-21:
Wow, your designs are really great. The Voronoi on Srf icon is better than the one from Grasshopper;) I'm still busy today, but tomorrow I will merge the current pulls. Thanks a lot already!!!

j8sr0230, 2022-11-21:
@luzpaz do you mean the Import Step node?

MisterMakerNL, 2022-11-21:
I made some icons myself:
https://mistermaker.gumroad.com/l/Linkstage_Icons
I also used Grasshopper couple of months ago for a job, and I use Sverchok and other nodes in Blender. So surely I watch this :)

j8sr0230, 2022-11-24:
@Emilien-Etadam Congratulations, your new icons are online :) Thanks for your efforts!

Emilien-Etadam, 2022-11-24:
Great ! It's only the beginning !

Emilien-Etadam, 2022-11-25:
[image]

Do you prefer I Pull each time or wait for more icons ?

j8sr0230, 2022-11-25:
Once again a great design :) I think it's enough if you make a pull every 4 to 5 icons. By the way, do you already know our FreeCAD Nodes topic in the forum? There I always publish the newest nodes. You can find it here: [https://forum.freecadweb.org/viewtopic.php?f=24&t=71651&start=50](https://forum.freecadweb.org/viewtopic.php?f=24&t=71651&start=50).

luzpaz, 2022-11-28:
@Emilien-Etadam would be great to see a PR for your work :stuck_out_tongue_winking_eye:

Emilien-Etadam, 2022-11-28:
@j8sr0230  Yes I keep an eye on the topic, thanks.
@luzpaz  I will PR (pull request ?) every 4 icons.

luzpaz, 2022-11-28:
@Emilien-Etadam Yea, no worries. We all went through our git learning phase (I still am :rofl:). 

What you can do is reference this PR with each PR you make by adding this PR number `#83` to the commit summary. When you're ready to close this PR you can add `Fixes #83` to the commit summary and it will automatically close when your PR is merged.

Emilien-Etadam, 2022-11-29:
@luzpaz  I don't understand anything, sorry...  why I should do this ?

Emilien-Etadam, 2022-11-29:
[image]
New Icons

luzpaz, 2022-11-29:
@Emilien-Etadam See https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

Emilien-Etadam, 2022-11-30:
@luzpaz I can't do this : 
[image]

luzpaz, 2022-11-30:
You're not. I didn't clarify myself enough. 

Make a PR and in the summary just mention this issue by it's number preceded with a hashtag like this:  `#83` (without the formatting of course, it's enabled for emphasis here). This will notate this issue (the one we are discussing at this moment) and reference that the commit you made is linked to this ticket. When the time comes that you want to close this ticket, you can do so by making a PR that has `Fixes #83` in the commit summary. In turn, when it is merged, this issue will automatically close. 

Does that make sense? (Sorry if I complicated things for you.)

Emilien-Etadam, 2022-12-05:
@j8sr0230  Why did you removed the icons here ?
[image]

j8sr0230, 2022-12-07:
I was not sure from the beginning if the many icons would overload the node graph.  Basically I aim for a clean clear layout like Blenders Geometry Nodes editor, but would let the community change my mind :)

PR #117 (2022-12-07): New icons #83

[image]

Fixes #83
closed 2022-12-07 by j8sr0230