j8sr0230/Nodes #102: Crv to face node error

Emilien-Etadam, 2022-11-30:
I dont understand how this node is working : 
[image]

Emilien-Etadam, 2022-12-05:
I've tried the fix on my fork and it's working fine now ! Thanks, @FlachyJoe

Emilien-Etadam, 2022-12-05:
This node is not working as I expected : 
I expected :
[image]
I get : 
[image]

It's not the same process as grasshopper ?

FlachyJoe, 2022-12-05:
@Emilien-Etadam Crv to Face results in two faces, it needs a not-yet-existent Merge node to combine the two curves in a unique one.

Emilien-Etadam, 2022-12-06:
Ok can't wait for it ! 👍

j8sr0230, 2022-12-06:
I put the Boundary Surface node on my list ;)

commit 7ac804b2 (2022-12-07): Merge pull request #111 from FlachyJoe/fix_102

Fix #102 Crv to face node error
commit cdf4bf59 (2022-12-07): Fix #102 Crv to face node error
closed 2022-12-06 by j8sr0230