j8sr0230/Nodes #119: Round number

Emilien-Etadam, 2022-12-07:
[image]

I m starting to play with the nodes but I'm stuck on a problem, I cant round a number to avoid this : 
[image]

closed 2022-12-08 by Emilien-Etadam

FlachyJoe, 2022-12-08:
Hi,
the required math functions can be done with a PyScript node (Keep in mind In and Out are lists).
[image]

Emilien-Etadam, 2022-12-09:
Ok now I see how it's works. Thanks.