j8sr0230/Nodes #123: Version 0.1.32 doesn't add extension to file name, subsequent open doesn't see the file.

macdroid53, 2022-12-09:
Version 0.1.32 doesn't add extension to file name, subsequent open doesn't see the file.

[code]
OS: Debian GNU/Linux 11 (bullseye) (XFCE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31277 (Git)
Build type: Release
Branch: master
Hash: 0d92058fb814671b4e10d679f38bc4dd5ffd6032
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * kicadStepUpMod 10.16.5
  * Manipulator 1.4.9
  * PrefExtras
  * QuickMeasure 2022.10.28
  * MyDefaults 2022.9.15
  * ZolkoBase 0.0.1
  * DynamicData 2.46.0
  * fasteners 0.4.25
  * Nodes 0.1.32
  * MacMechX 2022.9.15
  * ExtremeProDark 2.6.4
  * ProDarkThemePreferencePack 1.0.0
  * fcgear 1.0.0
  * plugins
  * Pyramids-and-Polyhedrons
  * Defeaturing 1.2.0
  * Assembly4 0.12.4
  * BaseTest 0.0.1
  * OpticsWorkbench 1.0.9
  * workfeature-macro
  * sheetmetal 0.2.59
  * A2plus 0.4.60f
  * CurvedShapes 1.0.4
  * BasicMech 0.0.1
  * Dracula 0.0.2
  * Curves 0.6.4
  * workfeature
  * toSketch 1.0.1
  * lattice2 1.0.0
[/code]

j8sr0230, 2022-12-09:
Thank you for your report. Did save and load work with the previous version or is this still the old problem?

macdroid53, 2022-12-09:
Save and load work. But, save does not add .json to the name. Then, subsequent attempts to open the file, with the default filter "Graph", do not show the file.
If the "All files" filter is selected the file is shown and will open fine.
This is not like before where the file would simply not open and throw a bunch of errors.

macdroid53, 2022-12-09:
Yes. Or even add the extension outside FreeCAD.

j8sr0230, 2022-12-09:
Interesting that windows and mac users do not have this error. For me, this is the case with GMSH. I always have to specify the extension... In the next update I'm gong to try this PySide Methode: `QFileDialog.setDefaultSuffix(suffix)`

luzpaz, 2023-01-08:
What's left here?

luzpaz, 2023-03-05:
@j8sr0230 bump