j8sr0230/Nodes #126: Unable to load node workbench

Nsikak1, 2022-12-12:
it says "NOTICE: Nodes content item Nodes does not support this version of FreeCAD, so is being skipped
"
And it doesn't show up in the node lists both from the addon manager and github clone directly.

[code]
OS: Ubuntu 22.04.1 LTS (LXQt/Lubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.
Build type: Release
Branch: unknown
Hash: 22f524c0c755003c12234fab1bad6a9f696b661a
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Nigeria (en_NG)
Installed mods: 
  * Render 2022.3.0
  * Nodes 0.1.34
[/code]

j8sr0230, 2022-12-12:
Alternatively you can install the workbench manually. You can find an instructions here: [https://wiki.freecadweb.org/How_to_install_additional_workbenches](https://wiki.freecadweb.org/How_to_install_additional_workbenches). But in addition you need to install the external Python modules `qtpy` and `awkward` in your FreeCAD environment.

Nsikak1, 2022-12-13:
@j8sr0230 so if it cant open in < 0.21 how do you get it to work on linkstage?(i have both qtpy and awkward installed). BTW what i am using is the latest version available on the ppa

Nsikak1, 2022-12-13:
@Emilien-Etadam i figured, i had to change the freecad admin tag for it to load is there any feature that prevents you from supporting a stable branch not every one has the luxury of using the latest greatest weekly build