j8sr0230/Nodes #129: B-spline surface ala microelly ?

parnasist3, 2023-02-02:
Any chance microelly's easy-editing b-spline surface can be done with your node version?

The killer feature would be the common sketch shared by the face portrayed in microellys video and another face adjacent to the first one (not pictured in his video).

I.e. both faces/surfaces share the same parent which means both surfaces are ruled/determed by 1) the sketch
2) the "tape" [as microelly call it]

j8sr0230, 2023-02-20:
Thanks for your request and the links. I am always happy about new FreeCAD knowledge. As I'm sure you've read in the FreeCAD forum, I'm currently working on a major revision of the Node Editor, which is mainly to improve the Node layout. Likewise, I want to cleanly implement the input widgets that were originally planned but then discarded. There should also be widgets for dircect point or vector input via the FreeCAD GUI, which would allow direct manipulation of BSpline surface control points. The creation of such surfaces is done by FreeCAD's OCC binding anyway and is already no problem, as far as I know...

michacassola, 2023-02-22:
Sorry to go off topic, but do you know how long you will probably take? Half a year? A year? More? I will not hold you to it, just to get a feel when this awesome workbench will be even awesomer 😄 . This will be the final nail in the Rhino/Grasshopper coffin for me.