j8sr0230/Nodes #131: Empty Nodes windows under macOS

pgilfernandez, 2023-04-07:
After installing in macOS the Nodes window appears blank or empty and it's not possible to do anything:

[image]

Any idea?
Thanks

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.32532 (Git)
Build type: Release
Branch: master
Hash: 20e44eba50e1365b9e7bb9c5c244afe96a627a28
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Nodes 0.1.36
  * Learning1
  * ExtremeProDark 2.7.8
```

origamifreak2, 2023-04-08:
I'm having this issue as well

```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.32532 (Git)
Build type: Release
Branch: master
Hash: 20e44eba50e1365b9e7bb9c5c244afe96a627a28
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Nodes 0.1.36
  * cadquery_module
```

kocsissandor85, 2025-01-06:
@origamifreak2 @pgilfernandez My quick was fix to hit Cmd+N to create a new graph, then use a right click to add nodes.