j8sr0230/Nodes #135: Cannot import name 'rand' from 'scipy'

A-Nicoladie, 2024-09-20:
Hi !
after installing Nodes, I have this message in the FreeCAD output

```
Unable to add node from "/home/xxxxxx/.local/share/FreeCAD/Mod/Nodes/./nodes/spatial/spatial_voronoi_on_sld.py": cannot import name 'rand' from 'scipy' (/tmp/.mount_FreeCAbS6cg1/usr/lib/python3.11/site-packages/scipy/__init__.py)
```

I tried to downgrade SciPy `pip install "scipy<1.12.0"` like mentioned in this issue (https://github.com/petercorke/RVC3-python/issues/16#issuecomment-2038741218), but it doesn't work...

Thanks for your help.

--
Linux Mint 21.3
FreeCAD 1.1.0dev (38768)
Python 3.10.12

havoc-dlt, 2024-09-30:
In [my pull request](https://github.com/j8sr0230/Nodes/pull/134) I specifically included a commit to 'solve' that issue:

This one:
https://github.com/j8sr0230/Nodes/pull/134/commits/26d1c122807cd32b9d92cb42ae834b8cedd46679

j8sr0230, 2024-10-04:
I have merged the pull. Should work again. Thank for the work!

closed 2024-10-04 by j8sr0230