j8sr0230/Nodes #138: Wire to face node

Tungee, 2025-01-09:
Hi there ,
in one in the examples in the Freecad forum there is a voronoi structure ( https://forum.freecad.org/viewtopic.php?t=71651&start=30 ). In the node tree there is a FromWire node, but in the latest nodes (not nodelink!) workbench there is not a such node.....
I know that nodes got structured and renamed in the meantime, but how can we achieve the same result without the FromWire node?! 
Best Regards

DaniloMitrovic, 2025-01-21:
It seems previous version worked with wires (like polylines) that then are converted into brep faces. Current version returns points, that seem to be ordered. Can you try making those points into curve:polyline and then surface: surface from curves?

Tungee, 2025-01-23:
> It seems previous version worked with wires (like polylines) that then are converted into brep faces. Current version returns points, that seem to be ordered. Can you try making those points into curve:polyline and then surface: surface from curves?

Hi Thank you Danilo!